Bodmer/TFT_ST7735 #1: The display pixel mapping is shifted, the fillScreen does not erase the whole screen

JiriBilek, 2017-02-23:
Hi,
I have 1.8 SPI TFT Display (https://www.aliexpress.com/item/New-Stylish-Hot-Sale-1-8-inch-128-x-160-Pixels-For-Arduino-TFT-LCD-Display/32611950535.html). The ID test shows me it has ST7735 chip:

```
TFT driver register values:
===========================
Register 0x01: 0x00
Register 0x04: 0x5C89F0
Register 0x09: 0x610000
Register 0x0A: 0x08
Register 0x0B: 0x00
Register 0x0C: 0x06
Register 0x0D: 0x00
Register 0x0E: 0x00
Register 0x0F: 0x00
Register 0x2E: 0xFFFFFF
Register 0xDA: 0x5C
Register 0xDB: 0x89
Register 0xDC: 0xF0
===========================

Looks like driver chip is: ST7735 (based on datasheet ID)

```
It seems to me there may be a bad timing or so, because the area of the display where I can write is shifted 1 pixel down and 2 pixel right. It means that writing to the point [0, 0] in fact writes to the physical position [2, 1] on the display.
I tried to write to the negative coordinates with some sucess: drawFastHLine and drawFastVLine easily wrote to area with negative coordinates while drawPixel does not. I am writing this here to give more information, it's clear using the negative coords is stupid :)

My Arduino sketch (I am running Arduino Pro 5V 16MHz but only on 3.3V in order not to fry the TFT):

```
#include <TFT_ST7735.h>
#include <SPI.h>

TFT_ST7735 tft = TFT_ST7735();  // Invoke library, pins defined in User_Setup.h

void setup(void) {
  tft.init();
  tft.fillScreen(TFT_BLACK); 
  tft.drawFastHLine(-2,-1,128,TFT_BLACK);  // all 3 lines draw ok
  tft.drawFastVLine(-2,-1,160,TFT_BLACK);
  tft.drawFastVLine(-1,-1,160,TFT_BLACK);
}

void loop() {
  tft.drawPixel(0,0,TFT_WHITE);  // draws ok
  tft.drawPixel(-1,0,TFT_WHITE);  // doesn't draw
  delay(200);
  tft.drawPixel(0,0,TFT_BLACK);  // draws ok
  tft.drawPixel(-1,0,TFT_BLACK);  // doesn't draw
  delay(200);
}
```

Thanks,
Jiri

Bodmer, 2017-02-24:
Have a look and the User_Setup.h file inside the library folder and try each tab colour option.

The drivers are wired to the screen pixels in different ways, some have a 1 or 2 pixel offset. Once the right tab colour option is selected the random pixel border should disappear. 

Unfortunately the screen protector tab colour is no longer a reliable indicate of how the driver is wired to the TFT pixels, so some experimentation to find the right option is needed.

Change the file, save, recompile and upload to test. Repeat for each option until the screen looks right.

Post back if that does not work for you.

Bodmer, 2017-02-24:
Try the INITR_BLACKTAB option first, then the REDTAB

JiriBilek, 2017-02-24:
Hi,
thanks a lot. I should have played with the settings more. My display has a black tab, but the sketch works on TAB_COLOUR INITR_REDTAB.
Now my testing sketch runs with no issues.

closed 2017-02-24 by JiriBilek