Bodmer/TFT_ST7735 #4: sd library problem with tft_st7735

soheils2, 2017-10-19:
hey 
u know i want to print bmp and raw files with your library
 but when i use that sketch for my ST7735 
2 errors hapend:

1-
```

 D:\arduino\Arduino\portable\sketchbook\libraries\TFT_ST7735-master/TFT_ST7735.h:348:12: error: 'void TFT_ST7735::setWindow(int16_t, int16_t, int16_t, int16_t) is private

     void   setWindow(int16_t x0, int16_t y0, int16_t x1, int16_t y1);

```

i solve this error with replacing "tft.setWindow()" with "tft.setAddrWindow()" and it compiled succesfully

2- i have already using "Adafruit_ST7735" library and [this example works fine](https://github.com/adafruit/Adafruit-ST7735-Library/blob/master/examples/soft_spitftbitmap/soft_spitftbitmap.ino) but u know! the library is too slow for drawing a bitmap!!.
it is almost the same as [your sketch](https://github.com/Bodmer/TFT_HX8357/tree/master/examples/Draw_SDCard_Bitmap)  .

but surprisingly!
when i upload your code the "SD.open(filename)" returns null and i get " File not found" error in terminal
PS:  the file is exist and the name is correct ! 
PS2: i even try your sample files  and get the same error !
PS3: using 
```
#include <SdFat.h>           // Use the SdFat library for the Due
SdFat SD;  
```
is the same and i just get " File not found" error in terminal :(

all things are the same i dont know what happening !
 can you help me to fix that?

this is the code i edit for ST7735 :  
```

// On the SD card use 24 bit color BMP files (be sure they are 24-bit!)
// There are examples images in the sketch folder.

// MS Paint can be used to crop, resize and flip and save images in 24bit BMP format

// Updated 31/3/15 to include new drawRAW() function that draws raw 16 bit images.
// Raw images are created with UTFT library tool (libraries\UTFT\Tools\ImageConverter565.exe)
// Save as .raw file as this can be easily piped to the TFT pushColors() function
// without any tedious byte swapping etc
 
#include <TFT_ST7735.h>        // Hardware-specific library
TFT_ST7735 tft = TFT_ST7735(); // Invoke custom library

// SD Card library, comment out next line if the SdFat library is used

#include <SD.h>

// We can use the SdFat library instead and it will be a bit faster, uncomment next 2 lines
//#include <SdFat.h>           // Use the SdFat library for the Due
//SdFat SD;                    // Permit SD function call compatibility with no sketch changes :-)

// SD chip select
#define SDC_CS 4

// These are used when calling drawBMP() function
#define BU_BMP 1 // Temporarily flip the TFT coords for standard Bottom-Up bit maps
#define TD_BMP 0 // Draw inverted Top-Down bitmaps in standard coord frame

/***************************************************************************************
** Function name:           setup
** Descriptions:            To set things up
***************************************************************************************/

void setup()
[... 232 lines cut ...]
/***************************************************************************************
** Function name:           Support functions for drawBMP()
** Descriptions:            Read 16 and 32-bit types from the SD card file
***************************************************************************************/

// BMP data is stored little-endian, Arduino is little-endian too.
// May need to reverse subscript order if porting elsewhere.

uint16_t read16(File& f) {
  uint16_t result;
  ((uint8_t *)&result)[0] = f.read(); // LSB
  ((uint8_t *)&result)[1] = f.read(); // MSB
  return result;
}

uint32_t read32(File& f) {
  uint32_t result;
  ((uint8_t *)&result)[0] = f.read(); // LSB
  ((uint8_t *)&result)[1] = f.read();
  ((uint8_t *)&result)[2] = f.read();
  ((uint8_t *)&result)[3] = f.read(); // MSB
  return result;
}

```

misi89, 2018-06-19:
I am facing with the same issue, but it shows image with the adafruit lib

misi89, 2018-08-19:
Thank you. Now it is loading the files, but they are upside down. If I change BU_BMP to TD_BMP it is still not corrext. Also it rotates the printed strings and numbers, but the position X and Y still according the original rotation.