Bodmer/TFT_ST7735 #6: TFT and SD do not work together

P123GT, 2018-03-01:
Hi,
first thanks for the great libs for the TFTs. I compared them with the ones from Adafruit but yours are much better!
But here is the problem:
I have the small 1.8" TFT with the ST7735 here (the china red one).
I have connected them to my Mega and have put a 74HC4050 in between to come down to 3.3V on the SCK, MOSI, DC, Reset, TFT_CS, SD_CS.
I use hardware SPI and configured the User_Setup.h.
Based on your example I made a sketch to show me some data from analog input on TFT only: works good.
Based on some SD example I made a logging function for SD-Card only: works good.
But if put the two together the TFT flashes and freezes.
Why?
René

`
#include <TFT_ST7735.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>
#include <SD.h>

TFT_ST7735 tft = TFT_ST7735();  // Invoke library, pins defined in User_Setup.h

#define TFT_GREY 0x5AEB // New colour

// Rotation Festlegen
byte Rotation = 4;

// Helligkeit festlegen
int led = 9;           // the PWM pin the LED is attached to
int brightness = 120;  // how bright the LED is (0-255, max. 168!)

// char array to print to the screen
char sensor1Printout[5];
char sensor2Printout[5];

// change this to match your SD shield or module;
const int SD_chipSelect = 53;

void setup(void) {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);

  Serial.print("\nInitializing SD card...");
  // we'll use the initialization code from the utility libraries
  // since we're just testing if the card is working!
  if (!SD.begin(SD_chipSelect)) {
    Serial.println("initialization failed. Things to check:");
    Serial.println("* is a card inserted?");
    Serial.println("* is your wiring correct?");
    Serial.println("* did you change the chipSelect pin to match your shield or module?");
    return;
  } else {
    Serial.println("Wiring is correct and a card is present.");
  }

  // declare pin 9 to be an output:
  pinMode(led, OUTPUT);
  // set the brightness of pin 9:
  analogWrite(led, brightness);

  tft.init();
  tft.setRotation(Rotation);  // Ausrichtung festlegen
  tft.setTextWrap(false);     // Weiterschreiben in neuer Zeile verhindern!

  // Fill screen with black
  // a background colour defined
  tft.fillScreen(TFT_BLACK);

  // Statischer Text der Größe 1:
  tft.setTextSize(1);

  // Set "cursor" at top left corner of display (0,0) and select font 2
  // (cursor will move to next line automatically during printing with 'tft.println'
  //  or stay on the line if there is room for the text with tft.print)
  tft.setCursor(0, 0, 2);

  tft.setTextColor(TFT_RED);
  tft.print("Speed:");

  tft.setCursor(64, 0);
  tft.setTextColor(TFT_YELLOW);
  tft.print("RPM:");

  tft.setCursor(0, 40);
  tft.setTextColor(TFT_GREEN);
  tft.print("Font2 - Z40:");

  tft.setCursor(0, 80);
  tft.setTextColor(TFT_BLUE);
  tft.print("Font2 - Z80:");

  tft.setCursor(0, 120);
  tft.setTextColor(TFT_BLUE);
  tft.print("Font2 - Z120:");
}

void loop() {

  // Read the value of the sensor on A1
  int sensor1Val = analogRead(A1);

  // Read the value of the sensor on A2
  String sensor2Val = String(analogRead(A2));

  // convert the reading to a char array
  sensor2Val.toCharArray(sensor2Printout, 5);

  // Set the font colour to be white with a black background
  tft.setTextColor(TFT_WHITE, TFT_BLACK); tft.setTextSize(1);

  // Set "cursor" at top left corner of display (0,0) and select font 4
  // (cursor will move to next line automatically during printing with 'tft.println'
  //  or stay on the line if there is room for the text with tft.print)
  tft.setCursor(0, 16, 4);
  // We can now plot text on screen using the "print" class

  char buff[10]; // Leerer Textspeicher

  //dtostrf(sensor1Val, 10, 0, buff); // dtostrf(double __val, char __width, char __prec, char *__s)
  sprintf(buff, "%4d", sensor1Val);
  tft.drawRightString(buff, 55, 16, 4); //drawRightString(char *string, int dX, int poY, int font)  tft.setCursor(64, 16);

  // Diese Version ist ok, aber rechtsbündig!
  tft.setCursor(64, 16);
  tft.print(sensor2Val);
  tft.print("      "); // 6Leerzeichen, um Reste zu löschen

  // Diese Version ist am besten geeignet, geht aber nicht mit Float!
  tft.setCursor(0, 56);
  sprintf(buff, "%4d", sensor1Val);
  tft.print(buff);
  tft.print("      "); // 6Leerzeichen, um Reste dahinter zu löschen

  tft.setCursor(64, 56);
  tft.print(sensor2Val);
  tft.print("      "); // 6Leerzeichen, um Reste dahinter zu löschen

  // Diese Version führt zu flickern
  tft.setCursor(0, 96);
  tft.print("                         "); // 25Leerzeichen, um Zeile zu löschen
  tft.setCursor(0, 96);
  tft.print(sensor1Val);
  tft.setCursor(64, 96);
  tft.print(sensor2Val);

  // Diese Version ist am besten geeignet, geht auch mit Float!
  float f_01 = sensor1Val / 1023.0 * 12.0;
  tft.setCursor(0, 136);
  dtostrf(f_01, 5, 1, buff); // dtostrf(double __val, char __width, char __prec, char *__s);
  tft.print(buff);
  tft.print("      "); // 6Leerzeichen, um Reste dahinter zu löschen

  // Diese Version ist Rechtsbündig, hat aber das Problem der Fragmente
  tft.drawRightString(sensor2Printout, 128, 136, 4); //drawRightString(char *string, int dX, int poY, int font)

  // wait for a moment
  delay(500);

  // make a string for assembling the data to log:
  String dataString = "";

  // read three sensors and append to the string:
  for (int analogPin = 0; analogPin < 3; analogPin++) {
    int sensor = analogRead(analogPin);
    dataString += String(sensor);
    if (analogPin < 2) {
      dataString += ",";
    }
  }

  // open the file. note that only one file can be open at a time,
  // so you have to close this one before opening another.
  File dataFile = SD.open("datalog.txt", FILE_WRITE);

  // if the file is available, write to it:
  if (dataFile) {
    dataFile.println(dataString);
    dataFile.close();
    // print to the serial port too:
    Serial.println(dataString);
  }
  // if the file isn't open, pop up an error:
  else {
    Serial.println("error opening datalog.txt");
  }

  // wait for a moment
  delay(500);
}
`

Bodmer, 2018-03-01:
Have you enabled transaction support?  See end of User_Setup.h file.

Also i set the SD chip select high at the start of setup(), then runt tft.init() then initialise the SD library.

P123GT, 2018-03-02:
No, it was disable.
But now it works! Great!

The other thing you mentioned I don't understand.

Bodmer, 2018-03-02:
Great that it is working.

The other things mention were ways to stop chip select conflicts during power up and before initalisation.  If two devices share an SPI bus it is good practice to make sure all chip selects are high before trying to initialise a SPI device. Some times the chip select lines are not pulled up so two devices get turned on at the same time, this may prevent initialising the SPI device.

If it works for you "as is" then that is fine.  The TFT_eSPI library sets the TFT chip slect high when the library object is created, not all libraries behave so well!

closed 2018-03-02 by Bodmer