Bodmer/TFT_ST7735 #10: How to implement the X-Y offset in the Library for an Adafruit 1.44" TFT?

cawena, 2018-07-20:
I added an INITR_144GREENTAB on user_setup.h
```
//#define TAB_COLOUR INITB
//#define TAB_COLOUR INITR_GREENTAB
#define TAB_COLOUR INITR_144GREENTAB
//#define TAB_COLOUR INITR_GREENTAB2
//#define TAB_COLOUR INITR_REDTAB
//#define TAB_COLOUR INITR_BLACKTAB

```
and added on TFT_ST7735.h
#define INITR_144GREENTAB 0x1
here
```
#define INITR_GREENTAB  0x0
#define INITR_REDTAB    0x1
#define INITR_BLACKTAB  0x2
#define INITR_GREENTAB2 0x3 // Use if you get random pixels on two edges of green tab display
#define INITB  0xB
#define INITR_144GREENTAB 0x1
```
and I changed also the siue of TFT
```
// Change the width and height if required (defined in portrait mode)
// or use the constructor to over-ride defaults
#define ST7735_TFTWIDTH  128
#define ST7735_TFTHEIGHT 128
```
on TFT_ST7735.cap
I added on the commandList:
```
  Rcmd2green144[] = {         // Init for 7735R, part 2 (green 1.44 tab)
      2,                      //  2 commands in list:
      ST7735_CASET  , 4    ,  //  1: Column addr set, 4 args, no delay:
      0x00, 0x00,             //     XSTART = 0
      0x00, 0x7F,             //     XEND = 127
      ST7735_RASET  , 4    ,  //  2: Row addr set, 4 args, no delay:
      0x00, 0x00,             //     XSTART = 0
      0x00, 0x7F },           //     XEND = 127
```
and after the command List I added the IF statement for INITR_144GREENTAB:
```
       else if (tabcolor == INITR_144GREENTAB)
       {
         commandList(Rcmd2green144);
         colstart = 2;
         rowstart = 3;
       }
```
but the issue is not fixed.
How can I change that?

THX

Bodmer, 2018-07-20:
Try choosing a different value for:

#define INITR_144GREENTAB 0x1

As 0x1 is already allocated to INITR_REDTAB

cawena, 2018-07-22:
OK, will try that. What is the meaning of this number?

cawena, 2018-08-07:
Still regarding the offset problem. I reloaded the original library.

I use your small test sketch to draw a frame around the TFT. First with Rotation "0". I get an offset of any pixels on the left side. When I use rotation "1" i get an offset on the top and left side. I also set the size of the TFT as it is written in the file TFT_ST7735.h.

```
#include <TFT_ST7735.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>

TFT_ST7735 tft = TFT_ST7735(128, 128); // Invoke library, pins defined in User_Setup.h

void setup(void) {
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
}

void loop() {
  tft.drawRect(0, 0, 127, 127, TFT_WHITE);
}
```
Any Idea how to fix that issue?

cawena, 2018-08-10:
I solved the problem more or less. At least I have the full width but the height is not right yet. I have made a few small additions. In TFT_ST7735.cpp at line 391:

```
       else if (tabcolor == INITR_144GREENTAB)
       {
         writecommand(Rcmd2green144);
         colstart = 3;
         rowstart = 2;
       }
```
and at 1555

```
else if(tabcolor == INITR_144GREENTAB){
         writedata(MADCTL_MY | MADCTL_MV | MADCTL_BGR);
         colstart = 3;
         rowstart = 2;
     }
```
a tft.drawRect (0,0,128,128, TFT_BLACK) is correct on the left, top, and right, but below the line is not drawn.

cawena, 2018-08-12:
i'm back to Adafruit Lib.

closed 2018-08-12 by cawena