Bodmer/TFT_ST7735 #12: Println, print or another method for text output by tft causes the Atmega2560 to stop working with CANopen Node.

cawena, 2018-08-06:
I have a small CAN Bus Programm just for check if the tft lib is working for my Project. This small Sketch run every 100ms a synch message to the node and the node response all seconds with a message. That is working fine. Serial println is showing me the Synch messages and also the responses.
When I try to put some message text on TFT, the Atmega2560 is stopped to communicate with the Node or can not read anymore from MOSI, MISO. I can send some Messages with the CANalyst2 to the Node and the node is still responding, but not the Atmega. Right now I have no idea how to find out where the problem is, or how to debug this in a good way.

in User_Setup.h:
I set  #define SUPPORT_TRANSACTIONS, because the MCP2515 is working with interrupts.

```
#include <SPI.h>
#include "mcp_can.h"
#include <Timer.h>
#include <TFT_ST7735.h

#define SPI_CS_PIN 31
#define TFT_CS 49
#define TFT_DC 48
#define TFT_RST 47
#define LITE 46

MCP_CAN CAN(SPI_CS_PIN);
Timer t;
TFT_ST7735 tft = TFT_ST7735(128, 128);

uint8_t len = 0;
unsigned char buf[8];
unsigned char flagRecv = 0;
const byte interruptPin = 21;
unsigned long current = 0;
unsigned long voltage = 0;

// PDO TX 280 Record
union {
  byte buf[8];
  struct {
    unsigned long current;
    unsigned long voltage;
  } rx;
} pdo2;

void setup() {
  Serial.begin(115200);

  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), MCP2515_ISR, FALLING); // start interrupt

  while (CAN_OK != CAN.begin(CAN_500KBPS)) {
    Serial.println("CAN BUS Shield init fail");
    Serial.println(" Init CAN BUS Shield again");
    delay(100);
  }
  Serial.println("CAN BUS Shield init ok!");

  tft.init();
  tft.setRotation(1);  // landscape
  clearScreen();

  t.every(100, sendHeartbeat);
}

void MCP2515_ISR() {
  flagRecv = 1;
}

void clearScreen() {
  tft.fillScreen(TFT_BLACK);
  setDefaults();
}

void setDefaults() {
  tft.setTextWrap(true);

  // Set "cursor" at top left corner of display (0,0) and select font 2
  tft.setCursor(0, 0, 2);

  // Set the font colour to be white with a black background
  tft.setTextColor(TFT_WHITE, TFT_BLACK);

  // set text size multiplier to 1
  tft.setTextSize(1);
}

void sendHeartbeat() {
  Serial.print("SEND SYNCH: ");
  Serial.println( millis());
  unsigned char msg[] = {0, 0, 0, 0, 0, 0, 0, 0};
  CAN.sendMsgBuf(0x80, 0, 0, msg);
}

void sendNMTMsg(uint16_t canId, uint8_t sendLen, uint8_t *msg) {
  _sendMsg(canId, sendLen, msg);
}

void sendMsg(uint16_t canId, uint8_t sendLen, uint8_t *msg) {
  _sendMsg(canId + 0x12, sendLen, msg);
}

//  '0' means standard frame. '1' means extended frame.
void _sendMsg(uint32_t canId, uint8_t sendLen, uint8_t *msg) {
  CAN.sendMsgBuf(canId, 0, sendLen, msg);
}

void setBatteryToOperationMode() {
  uint8_t opMode[] = {0x01, 0x12};
  sendNMTMsg(0x0, 2, opMode);
}

void loop() {
  t.update();

  if (flagRecv) {
    flagRecv = 0;

    while (CAN_MSGAVAIL == CAN.checkReceive()) {
      memset(buf, 0, sizeof(buf));
      CAN.readMsgBuf(&len, buf);
      uint16_t canId = CAN.getCanId();
      uint16_t canIdRaw = canId - 0x12;

      Serial.print("ID: 0x");
      Serial.print(canIdRaw, HEX);
      Serial.print("\t");
      for (int i = 0; i < len; i++) {
        Serial.print(buf[i], HEX);
        Serial.print("\t");
      }
      Serial.println();

      if (canIdRaw == 0x700 && buf[0] == 0x7F) {
        setBatteryToOperationMode();
      }

      if (canIdRaw == 0x280) {
        pdo2.buf[0] = buf[0];
        pdo2.buf[1] = buf[1];
        pdo2.buf[2] = buf[2];
        pdo2.buf[3] = buf[3];

        pdo2.buf[4] = buf[4];
        pdo2.buf[5] = buf[5];
        pdo2.buf[6] = buf[6];
        pdo2.buf[7] = buf[7];
      }
    }
  } else {
    char formattedCurrent[7];
    // tft.fillScreen(TFT_BLACK);
    sprintf(formattedCurrent, "%d%mA", pdo2.rx.current);
    tft.setTextSize(2);
    tft.setCursor(0, 0);
    tft.print((int)pdo2.rx.current);

    // tft.print(pdo2.rx.current);
    // tft.setTextColor(TFT_BLACK, TFT_BLACK);
    // tft.drawString("4000mA", 10, 20, 1);
    // tft.setTextColor(TFT_WHITE, TFT_BLACK);
    // tft.drawNumber("4000mA", 10, 20, 1);
}

```

Some idea?

Bodmer, 2018-08-07:
If the CAN bus library is using interrupts then it is probably grabbing the SPI bus while the TFT is being updated and it appears the CAN bus library itself does not support transactions.

You could try disabling interrupts before any TFT screen updates and enabling them afterwards. This may help isolate the problem.

cawena, 2018-08-07:
it's not necessary to use interrupts with the CAN Library but it's the best way to not lost some CAN Data. I will try to switch on and off during writing to tft. I think that this is an ON and OFF game with the CAN bus Interrupt and I might lose data as a result. But I will try.

closed 2018-08-07 by cawena