Bodmer/TFT_ST7735 #14: drawFloat problem

scrungy, 2019-05-09:
There is a corner case error.
If you pass in a number say 3.1415    and set dp = 5, the floating point number does not print the fractional part.  This is because it relies on the variable i, which was set earlier.
for (i=0;i<dp;++1) rounding /=10.0;
leaves i = the number of characters to the right of the decimal point, in this case 5.
Then when it gets to this line:
while((i<dp) && (digits <9)) the test fails, because dp and i are equal.
the solution is to put
i=0; in front of the while.  Discovered when porting this to a PIC24.

Bodmer, 2019-05-09:
This appears to be an issue introduced during the porting as the original code has different scopes for the i variable. See [here](https://github.com/Bodmer/TFT_ST7735/blob/master/TFT_ST7735.cpp#L2083) and [here](https://github.com/Bodmer/TFT_ST7735/blob/master/TFT_ST7735.cpp#L2121).

scrungy, 2019-05-10:
Well, I guess that is the problem with the variable.  Thank you for 
pointing that out.  Allocating a variable and assigning it a value like 
that down inside of a function, is really
not a good thing.  Code should be readable, it is for the programmer.  
Variables should all be created and grouped together.

I'm old school and absolutely HATE it when programmers initialize and 
create a variable locally like that.  They HAVE to be at the top of the 
function, so you can see where they are.
I also trace most functions at the assembly language level.  So I move 
them.  I'm a bare metal type embedded programmer and have been doing it 
since 1977.  My first computer was a 1 mhz 6800 with 256 BYTES of ram 
and 512 BYTES of rom that held Mikbug.  300 baud serial, baud rate 
adjusted via two pots, that is how old school I am.   :)  Currently I 
work for a medical
device manufacturer writing code for a hand held micro current TENS unit...

Porting your library to the PIC has been a real bear.  The XC16 compiler 
is NOT a C++ version of the gnu compiler, even though it is the full gnu 
compiler with microchip tweaks.

So I have had to make extensive changes all over the place, converting 
variables that were normally in the class to standalone, taking 
overloaded functions and creating new names
for them etc.  Had to change the machine code, but then once you do that 
and try to run it at 247k, the chip selects and data doesn't line up on 
the logic analyzer, so I pretty much had
to hack it hard.

While I rewrote the spiwait16, etc with the correct assembly code for 
the PIC24, I wound up getting rid of them and using a simpler macro.   
Does string pasting.
The compiler even on optimzation 1 (which is the max that the free 
version can do) was generating horrible code.  This is 4 cycles tops.
and yes, it is much slower than it needs to be, but I had to get it 
working first.

*then* go back and optimize.  To make things like chip select, and 
command toggles line up on the Salae logic analyzer, I need all the spi 
data transmits to hang UNTIL THE SHIFT REGISTER IS EMPTY....
I think a lot of this would not have been necessary, if I had realized 
that the mcc code generator had used enhanced mode.  That really caused 
lots of grief.  Once I figured that out, then
it still would not work, and that turned out that the Polarity was set 
wrong.  It took weeks to figure all that out....
anyrate, when I get done, I will have a pure C module, for the PIC24 
family, with some routines pulled from other libraries merged into it.  
Specifically, the round gauge routine from sumtoy.

#define SPITBF 1 /*bit number in SPI1STATL */

#define xstr(s) str(s))
#define str (s) #s
#define hang_until_set (addr,bit) \
           asm volatile ("1:     btss   "xstr(addr) ",#" xstr(bit) "\n" \
                                    "       bra 1b \n ")
     hang_until_set (SPISTATL,SPITBF);

Now that I got it running, I am testing the functions, and will write 
some C wrappers to create something that is more logical. Because this 
is a non C++ compiler, you
can't inherit the properties of the print class, so  println does not work.

Once I get this all cleaned up and all the debug lines out, I will send 
it to you, if you give me your email address.....
This is my company email account, my personal one is knapper@realtime.net

-- 
There are TWO type of people in this world
1. those who can extrapolate from incomplete data.