Bodmer/TFT_ST7735 #16: Pixel Offset at Top and Left

McNugget6750, 2019-10-16:
While I was bringing up a battery monitor that worked fine with one display on my breadboard and then didn't with essentially the same display on my PCB, I came across the thread here, which was extremely helpful!
I want to use your libraries because they are insanely fast! However, I don't get the same drawing reliability with your library, compared to the slow Adafruit lib:
[image]
As you can see, the image looks fine using the Adafruit lib.

However, when I use yours, the entire screen is shifted by 1px horizontally and 2px vertically:
[image]
Everything else seems to work just fine so I suspect there might be bug in one of the display driver components on your end.

Please let me know if and how I can help investigate since I am really impressed with the performance of this lib! Thanks for putting in all the work!

Cheers,
Timo

Bodmer, 2019-10-16:
Different display boards use different offsets. The default display uses an offset.

Try using the [setup](https://github.com/Bodmer/TFT_ST7735/blob/master/User_Setup.h#L7-L11):
#define TAB_COLOUR INITR_GREENTAB
option instead of
#define TAB_COLOUR INITR_GREENTAB2

McNugget6750, 2019-10-18:
Thanks a lot! I tested all and 
`#define TAB_COLOUR INITR_BLACKTAB`
seems to have done the trick.

closed 2019-10-18 by McNugget6750

Bodmer, 2019-10-18:
Gret, thanks for letting me know.