Bodmer/TFT_ST7735 #20: ESP32-PICO-D4 ST7735 180x60 deep sleep

biccius, 2020-10-17:
Hi,

I'm programming a LilyGo T-Wristband with Arduino core (no esp-idf) using ESP32-PICO-D4.
A ST7735 0.96" 160x80 TFT screen is connected using SPI and TFT_eSPI is the graphical library used to manage the display.
A TP button is used to wake up the chip from deep sleep mode.
Schematics and more informations are at the end of this post.

My purpose is to try to:

a. init display at the start then fill the screen with a color A
b. send the esp32 to deep sleep
c. wake up with button press the esp32 and redraw the display with color A avoiding to re-initialize the display cause it's already set
d. then change in progression colors to B, C , etc..

On "c" phase i don't want to re-initialize the display everytime i wake from a deep sleep because is a slow operation.
I assume the display is already initialized and just wake it up from sleep as fast as possible.
So i'm trying with this code to keep the RESET pin high (gpio_hold_en) and switch off the display (DISPOFF) and sending it to sleep (SLPIN)

```

#include <TFT_eSPI.h>            
#include <SPI.h>
#include <WiFi.h>
#include <Wire.h>
#include "sensor.h"
#include "esp_adc_cal.h"

#define TP_PIN_PIN          		33		// wakeup pin
#define TP_PWR_PIN 			25		// power for the touch button

TFT_eSPI tft = TFT_eSPI();       // Invoke library, pins defined in User_Setup.h
RTC_DATA_ATTR int boots = 0;

void setupTFT()
{
  tft.init();
  tft.setRotation(1);
  tft.setSwapBytes(true);
}

int color = 0x5555;

void Wake()
{

    pinMode(TFT_CS, OUTPUT);
    pinMode(TFT_DC, OUTPUT);
    pinMode(TFT_BL, OUTPUT);

    digitalWrite( TFT_BL , HIGH);	//backlight on

    SPI.begin(TFT_SCLK, TFT_MISO, TFT_MOSI, -1);

    tft.writecommand(ST7735_DISPON);	// display on
    delay(150);
    tft.writecommand(ST7735_SLPOUT);	// display sleep off
    delay(150);
}

void Sleep()
{
  tft.writecommand(ST7735_SLPIN);  // display sleep on
  delay(150);
  tft.writecommand(ST7735_DISPOFF);  // display off
  delay(200);

  digitalWrite(GPIO_NUM_26, HIGH);  	// keep GPIO_26 high state after deep sleep reset
  gpio_hold_en(GPIO_NUM_26);		// display is preserved
  gpio_deep_sleep_hold_en();

  digitalWrite( TFT_BL , LOW);  //backlight off
  SPI.end();	// spi communication stopped

  esp_sleep_enable_ext1_wakeup(GPIO_SEL_33, ESP_EXT1_WAKEUP_ANY_HIGH);
  esp_deep_sleep_start();
}

void setup(void)
{
	  pinMode(TP_PIN_PIN, INPUT);
	  pinMode(TP_PWR_PIN, PULLUP);
	  //! Must be set to pull-up output mode in order to wake up in deep sleep mode
	  digitalWrite(TP_PWR_PIN, HIGH);

	  Serial.begin(115200);

	  if(boots == 0) //Run this only the first time
	  {
		  Serial.println("first boot");
		  setupTFT();
		  tft.fillScreen(color);
		  color+=0x1000;
	  	  boots ++;
	  }
	  else
	  {
		  Serial.println("wake");
		  Wake();
		  tft.setRotation(1);
		  tft.setSwapBytes(1);
		  delay(3000);
		  while (1)
		  {
			  Serial.println("changing color");
			  tft.fillScreen(color);
			  color+=10;
			  delay(10);
		  }
	  }//if boots
}

void loop()
{
	delay(5000);
	Serial.println("sleep");
	Sleep();
}
```

Code seems to work properly but in "d" phase i get the display not full filled by a new color like in figure:
https://i.imgur.com/LYRMzat.jpeg

Is there something wrong? 
What can be the cause and how to fix it?
Any code improvements?

Thank you

**PIN DEFINITION**

TFT_MOSI 	19
TFT_SCLK 	18
TFT_CS 	5
TFT_DC 	23
TFT_RST 	26
TFT_BL 	27
Touchpad 	33
Touchpad Power 	25

**RESOURCES**

-LilyGo T-Wristband : https://github.com/Xinyuan-LilyGO/LilyGo-T-Wristband
-Schematics: https://github.com/Xinyuan-LilyGO/LilyGo-T-Wristband/blob/master/schematic/T_Wristband_lsm9ds1_20200306.pdf
-Issue description: https://github.com/Xinyuan-LilyGO/LilyGo-T-Wristband/issues/19

Bodmer, 2020-10-18:
Make sure the TFT_CS signal stays high otherwaise the display can get garbage commands. I suspect the display may need to be re-initialised upon wake up. Unfortunately I am unable to advise further.

biccius, 2020-10-18:
Thank you very much for the advice. I tried them both.

1. NOT WORKING: Added _gpio_hold_en(GPIO_NUM_5)_  in Sleep function to keep CS signal high: same result of my previous code. After the wake up the display is not filled completely. 

```
#include <TFT_eSPI.h>            
#include <SPI.h>
#include <WiFi.h>
#include <Wire.h>
#include "sensor.h"
#include "esp_adc_cal.h"

#define TP_PIN_PIN          33
#define TP_PWR_PIN 			25

TFT_eSPI tft = TFT_eSPI(80,160);       // Invoke library, pins defined in User_Setup.h

RTC_DATA_ATTR int boots = 0;
			  int color = 0x5555;

void setupTFT()
{
  tft.init();
  tft.setRotation(1);
}

void Sleep()
{
	digitalWrite( TFT_BL , LOW);
	tft.writecommand(ST7735_DISPOFF);
	SPI.end();
	digitalWrite(GPIO_NUM_5, HIGH);  // CS high after sleep
	gpio_hold_en(GPIO_NUM_5);

	digitalWrite(GPIO_NUM_26, HIGH);  //RESET high after sleep
	gpio_hold_en(GPIO_NUM_26);
	gpio_deep_sleep_hold_en();

  esp_sleep_enable_ext1_wakeup(GPIO_SEL_33, ESP_EXT1_WAKEUP_ANY_HIGH);
  esp_deep_sleep_start();
}

void setup(void)
{

	  Serial.begin(115200);

	  if(boots == 0) //Run this only the first time
	  {

		  pinMode(TP_PIN_PIN, INPUT_PULLDOWN);
		  pinMode(TP_PWR_PIN, PULLUP);
		  //! Must be set to pull-up output mode in order to wake up in deep sleep mode
		  digitalWrite(TP_PWR_PIN, HIGH);

		  Serial.println("first boot");
		  setupTFT();
		  tft.fillScreen(color);
		  color+=0x1000;
	  	  boots ++;
	  }
	  else
	  {
		  	Serial.println("wake");

		  	pinMode(TFT_BL, OUTPUT);
		  	digitalWrite( TFT_BL , HIGH);

		  	gpio_hold_dis(GPIO_NUM_5);		// CS hold disabled
		    gpio_hold_dis(GPIO_NUM_26);		// RESET hold siavled
[... 174 lines cut ...]

		  delay(3000);
		  while (1)
		  {
			  Serial.println("changing color");
			  tft.fillScreen(color);
			  color+=10;
			  delay(10);
		  }
	  }//if boots
}

void loop()
{
	delay(5000);
	Sleep();
}
```

It is not clear to me why method 1 is not sufficient, in any case method 2 seems to work.

Thanks for the suggestions. 
If there are no other improvements on the codes proposed for me you can close the issue.

Thank you very much.

biccius, 2020-10-27:
Unfortunately I was wrong.
The screen fill seems to work but other functions such as writing text for example do not work properly so unfortunately it seems that something needs to be reset and re-initialized.